Bully85/Sovol-SV04-Klipper #3: Is this good to use with sonic pad?

scuba1976, 2023-05-26:
Is this good to use with sonic pad? Is it possible to use all sv04 function on sonic pad?

Bully85, 2023-05-26:
For use this you must choose my repo. At the moment sonic pad do not support to change the repo sorry

scuba1976, 2023-06-05:
Dear friend, i tried to do all the steps with raspberry 3b+. but i cannot connect to the printer. which firmware i have to use? there are two file.
I just need to copy all .cfg in config?
thanks

scuba1976, 2023-06-05:
Hello, i need you help. I install everything on my sv04. i have GD32F303, but z axis doesn't work. what i make mistake?
thanks

Bully85, 2023-06-05:
Do you change the z sector in printer.cfg

scuba1976, 2023-06-06:
Hello, i'm sorry to bore you. I success in all the steps and i start to print. The problem is that, after z tilt calibration with right extruder, it seams to start printing with left extruder. Can you help me? thanks

closed 2023-06-12 by scuba1976

Funkelfetisch, 2023-10-09:
reopening this, anybody checked out:

https://github.com/smwoodward/Sonic-Pad-Updates/blob/main/root_access/Root

yet?