Bully85/Sovol-SV04-Klipper #5: Reflash to marlin possible ?

pleasanthill17, 2023-05-29:
I am considering flashing to Klipper. However since this not special repo, I am wondering if you can return to Marlin - reflash to Marlin in the future after trying Klipper ? Thanks for your work on this !

Bully85, 2023-05-29:
Yes every time is possible. You only must flash the bin and dwin set for marlin

animerunt, 2023-07-06:
Is there anything special that needs to be done to flash the screen to another update?
I successfully flashed johncarlsons firmware over the klipper firmware.
I'm attempting to flash the screen with his screen firmware too:
https://github.com/johncarlson21/SV04-Touchscreen/releases

My SD card is good and formatted to fat32, same as when I flashed your screen firmware. 
When I turn it on to update, it just goes to the klipper menu, which of course doesn't work now that the main firmware is back to marlin.

animerunt, 2023-08-18:
Sorry for the delay. I was able to flash the screen firmware after trying a different SD card. 
I had been busy and only just got around to trying this again. 
Thanks for your hard work! I will probably return in the future after it's updated more to try klipper again!