Bully85/Sovol-SV04-Klipper #11: Klipper issue with variables.cfg

CornBred89, 2023-07-14:
I get the following error when I launch mainsail for my SV04:

Internal error during connect: [Errno 2] No such file or directory: '/home/Klipper/printer_data/config/variables.cfg'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

How can I fix this? I have updated the printer.cfg file with all my corrected values from a previous Klipper install on my SV04.

Bully85, 2023-07-14:
The variables.cfg is missing

CornBred89, 2023-07-14:
I uploaded all of the configuration files, including the variables.cfg file, to mainsail before I got the error. I even tried including it in the printer.cfg file. That didn't work either.

CornBred89, 2023-07-15:
Nevermind, I figured it out. I have two instances of moonraker on my pi. So, I had to go into the misc.cfg file and update the file paths found in that config file.

Bully85, 2023-07-16:
Ok than you must rename the path