Bully85/Sovol-SV04-Klipper #13: Cura 5.3

Malvus524, 2023-07-23:
Maybe it's a stupid question but in the description you mention Cura 5.3 is currently not working.
Does this mean any newer version is affected as well?
Because at the moment I am using Cura 5.4 and I don't know where I should get any older version.
I don't want to configure it all and can't use it at the end.

Thanks for your help and the overall effort to provide this config to the community up front

Bully85, 2023-07-23:
Sorry at the moment I must work very hard in my main job I try to test it tomorrow

Bully85, 2023-07-23:
But you can import the profile an slice a stl. Than Post the gcode head. The problem of 5.3 is that this don't give the extruder park to Klippe start macro

Malvus524, 2023-07-23:
[image]

Malvus524, 2023-07-23:
What is missing now?

Malvus524, 2023-07-23:
If you send me pictures from the correct start and end g code and I add it to the Cura start and end g code it should be fine

Bully85, 2023-07-23:
The problem is the same extruder parms only -1 must be 1 or 0

Bully85, 2023-07-23:
Cura change the extruder Parma's.with 5.3 I search for a way that allow 1 startcode for all slicer

Bully85, 2023-07-23:
[image]

[image]

Malvus524, 2023-07-23:
I don't quite get it yet.
Mainly I want to use dual mode because even if I use only one extruder I simply select extruder 1 or 2 in Cura.
Can you send me a picture from the whole start g code (not just the post/prefixes in Cura) I need to get dual mode working please?

Funkelfetisch, 2023-10-08:
any update on this?
I have a creality sonicpad here and want to hook it up :D

Bully85, 2023-10-08:
If the sonically up to date with the origin repo? Than it will be work with my new release. Not without the original display

Bully85, 2023-10-10:
Das problem ist, dass ich noch nicht herausgefunden habe wie ich ab cura 5.3 die variable setzen muss, dass er übermittelt mit welchem extruder er anfängt im Dual Modus. Ich werde diese Woche mal auf 5.4 updaten und mich damit beschäftigen

andrezanna, 2024-08-03:
Hello, I was able to correctly print with cura 5.8 by usign this little script to change the start extruder
```
import re
from ..Script import Script

class ChangeStartExtruder(Script):

    def getSettingDataString(self):
        return """{
            "name": "ChangeStartExtruder",
            "key": "ChangeStartExtruder",
            "metadata": {},
            "version": 2,
            "settings":{}
        }"""

    def execute(self, data):
        
        incremented_extruder_nr = None

        print(data)
        for layer_number, layer in enumerate(data):
            extruder_result = re.search("START_PRINT EXTRUDER=-?(\d+)", layer)
            print(extruder_result)
            if extruder_result:
                extre_string=extruder_result.group().split('=')[1]
                print(extre_string)
                extruder_nr = int(extre_string)
                print(extruder_nr)
                incremented_extruder_nr = extruder_nr + 1
                print(incremented_extruder_nr)
                data[layer_number] = re.sub("START_PRINT EXTRUDER=-?(\d+)", "START_PRINT EXTRUDER="+str(incremented_extruder_nr), layer)
        
        return data

```

Print went perfectly fine

Malvus524, 2024-08-06:
How did you install this script?

andrezanna, 2024-08-06:
Follow the same procedure for the printMeshSize script, Just enable ChangeStartExtruder

Malvus524, 2024-08-06:
okay i have tried but it looks like this: 
[image]
 
That dosen't seem right

Malvus524, 2024-08-06:
it just says [current_extruder] no matter what extruder it should start with

andrezanna, 2024-08-06:
If you disable the script what value do you have there?

andrezanna, 2024-08-06:
And which klipper base profile do you use?

Malvus524, 2024-08-06:
its the same as with the script.

i don't get the question

andrezanna, 2024-08-06:
Strange, I was getting a different value.

I mean when you create a printer there are many sv04 klipper profile, which one did you use?

Malvus524, 2024-08-06:
I used the profile from this GitHub

andrezanna, 2024-08-06:
Ok, then can you try with this:

```
import re
from ..Script import Script

class ChangeStartExtruder(Script):

    def getSettingDataString(self):
        return """{
            "name": "ChangeStartExtruder",
            "key": "ChangeStartExtruder",
            "metadata": {},
            "version": 2,
            "settings":{}
        }"""
        
    def try_parse_int(s, base=10, val=None):
      try:
        return int(s, base)
      except ValueError:
        return val
    
    def execute(data):
        
        incremented_extruder_nr = None
    
        for layer_number, layer in enumerate(data):
            matchline = re.search("START_PRINT EXTRUDER([^\s]+)", layer)
            if matchline:
                elements=layer.split(" ");
                for chunk in elements:
                    extruder_result=re.search("EXTRUDER([^\s]+)",chunk)
                    if extruder_result:
                        extre_string=extruder_result.group().split('=')[1]
                        extruder_nr = try_parse_int(extre_string,val=-1)
                        incremented_extruder_nr = extruder_nr + 1
                        data[layer_number] = re.sub("START_PRINT EXTRUDER=([^\s]+)", "START_PRINT EXTRUDER="+str(incremented_extruder_nr), layer)
            
        return data

```