Bully85/Sovol-SV04-Klipper #16: trying to follow the instructions in readme.md, but config files unclear

Dweinbach, 2023-08-11:
Trying to install but no joy. 

first i tried the image but always got 404 from NginX

then I tried my working (desktop OS) pi image, and got alot farther, but the readme.md says "copy the config folder to printer_data". is this really correct - like all the files in the repo's config folder go into /home/[mynamehere]/printer_data ? Not into /home/[mynamehere]/printer_data/config ? I need more clarity.

RizeCrime, 2023-08-11:
the entire config folder needs the be copied into printer_data.
So you want to end up with "xyz/printer_data/config/printer.cfg" (and all the other config files ofc).

It's clearer under "Installation on a new OS", the other section needs to be updated.

Dweinbach, 2023-08-12:
I would suggest changing:
"Now, the files and folders from the "config" directory must be copied to your printer's directory on the Raspberry Pi. The default is "printer_data". If this is not the case for you, please adjust the paths in the config and .sh files."
to:
"Now, the entire "config" folder and its contents must be copied to your printer's directory on the Raspberry Pi. The default is "~/printer_data". If this is not the case for you, please adjust the paths in the config and .sh files."