Bully85/Sovol-SV04-Klipper #19: Filament Unload in KlipperScreen is missing UNLOAD_FILAMENT Macro

Dweinbach, 2023-08-16:
I just tried to use unload filament in KlipperScreen, and it says "Macro UNLOAD_FILAMENT not found" for both extruders.

Bully85, 2023-08-23:
I check this

2greatsecret, 2024-05-22:
The problem is still present.
It is also not preheated when you want to load or unload filament

2greatsecret, 2024-05-22:
I copied the following macro from Kingroon KLP1 into macros.cfg and adjusted it very slightly (the filament length when loading and unloading).
Works quite well, even if not as well as the KLP1

```
[gcode_macro LOAD_FILAMENT]
gcode:
    M104 S220
    M105    
    M109 S220
    G91 
    G1 E50 F300
    G90

[gcode_macro UNLOAD_FILAMENT]
gcode:
    M104 S220
    M105    
    M109 S220
    G91
    G1 E30 F3000
    G1 E-27 F9000
    M106 S255
    M104 S62
    M105    
    M109 S62
    G1 E-100 F300
    M106 S0
    M84
```