Bully85/Sovol-SV04-Klipper #23: Original Display

Malvus524, 2023-09-30:
In the info you say something about an orig Display:
Do you mean the Original display from sovol or from the raspberry pi?
I ask because i cant get the Display from sovol to work.

Bully85, 2023-09-30:
Which firmware do you install?

Malvus524, 2023-10-01:
I was am using your old firmware but I have the configs updatet to the new ones but I can't get it to work.
I have updated start end macros, macros and index mode

Bully85, 2023-10-01:
You must use the new firmware.bin all you need is in release 1.0.1

Bully85, 2023-10-01:
https://github.com/Bully85/Sovol-SV04-Klipper/releases/tag/1.0.1

Malvus524, 2023-10-01:
it still does not work. i have installed your new firmware on the mcu but i cant get the display to change the bin into a cur.
what am i doing wrong?

Bully85, 2023-10-01:
Which config fiele you use?

Bully85, 2023-10-01:
Do you change the repository?

Malvus524, 2023-10-01:
The config file is the old one but I didn't change the repository

Malvus524, 2023-10-01:
Do I have to install the new firmware to both the display and the MCU?

Bully85, 2023-10-01:
for printer install: https://github.com/Bully85/Sovol-SV04-Klipper/releases/download/1.0.1/firmware.bin
for display: https://github.com/Bully85/DGUS-SV04/releases/download/1.0.0/DWIN_SET.rar

change the repository: https://github.com/Bully85/Sovol-SV04-Klipper#if-you-use-kiauh

the config: https://github.com/Bully85/Sovol-SV04-Klipper/releases/download/1.0.1/config.with.and.without.Display.rar

than import your slicer profiel cura(https://github.com/Bully85/Sovol-SV04-Klipper/releases/download/1.0.1/Cura.Profile.include.startcode_md.rar) or Prusa (https://github.com/Bully85/Sovol-SV04-Klipper/releases/download/1.0.1/Prusaslicer.Profile.include.startcode_md.rar)

now all must run

Malvus524, 2023-10-01:
Thank you very much

scuba1976, 2023-10-15:
I install original display and it works. It is normal the there isn't "print" command?

Bully85, 2023-10-15:
Yes at the moment. You can't start a print with the display. At the moment only for temp and cancel and secure stop