Bully85/Sovol-SV04-Klipper #25: Unknown command:"SET_DUAL_CARRIAGE_MODE" and "Must home axis first: 0.000 0.000 0.200 [0.000]

scuba1976, 2023-10-13:
I updare klipper using this steps"https://github.com/Bully85/Sovol-SV04-Klipper#installation-on-an-existing-os"
Now, when i try to print in dual mode  it appears "Unknown command:"SET_DUAL_CARRIAGE_MODE" and "Must home axis first: 0.000 0.000 0.200 [0.000]" and the print don't start. can you help me?
[klippy.log](https://github.com/Bully85/Sovol-SV04-Klipper/files/12894522/klippy.log)

Bully85, 2023-10-13:
You don't have the correct config files use this https://github.com/Bully85/Sovol-SV04-Klipper/releases/tag/1.0.1

scuba1976, 2023-10-13:
I copied the file from config folder. I don't use original display. can you help me to put the correct config?

Bully85, 2023-10-13:
config/Sv04-works-with origin-klipper

You need Macro cfg idex cfg and start end macro cfg

scuba1976, 2023-10-13:
I copied all the config folder and file in printer data, but alwayes the same problem.
could be a problem with old 
prusaslicer config?

Bully85, 2023-10-13:
Yes you must change it too

Bully85, 2023-10-13:
In the path is a text file with new start code

scuba1976, 2023-10-13:
Thanks, now it works. I get still two message : Hyperlaps unknown and dgus_print_start error. can you help me any more?

lucwuyts, 2023-10-23:
I think dgus_print_start is a command/macro to show something on the original display.
dgus_print_end is in macros.cfg
dgus_print_start and dgus_print_end is in  start_end_macro.cfg

I removed this because i use a PAD7 as display.  Original display is removed on my machine.

Bully85, 2023-10-23:
You're right

Bully85, 2023-10-23:
> Thanks, now it works. I get still two message : Hyperlaps unknown and dgus_print_start error. can you help me any more?

Can you take a screenshot

scuba1976, 2023-10-23:
I got everything. Thanks to all

closed 2023-10-23 by scuba1976

Bully85, 2023-10-23:
Your welcome