Bully85/Sovol-SV04-Klipper #26: Must home axis first when resume after fillament runout

lucwuyts, 2023-10-20:
Fillament was runout , changed it, and pressed the resume button.
Now this message appears:
Must home axis first: 128.074 62.714 37.094 [142520.420]

Why is this?  The printer clearly knows where the head is.
And homing will collide with the unfinished print

Bully85, 2023-10-20:
Where do you press resume?

Bully85, 2023-10-20:
Can you try to reproduce it and make a video?

lucwuyts, 2023-10-20:
I press resume on the PAD 7 screen.
Will try to make a video.

lucwuyts, 2023-10-20:
First attempt: print continued as expected.   But this was now at the first layer.    I will try tomorrow with a simple tower at higher Z to repeat this.

Bully85, 2023-10-20:
Okay, I'm waiting for feedback

lucwuyts, 2023-10-21:
It seems after 2 hours, the bed and extruder heating stops, and when you press the resume button, this message appears.
Probably this is a bug in the PAD7  or Klipper itself?   
I attach 2 photo's.  The first showing the printer is paused and extruder has cooled down, second after pressing the resume button.
In the 2 hours between the sensor detecting no more fillament, i was always able to restart the print.

[image]

[image]

lucwuyts, 2023-10-21:
I found an example to add to the printer.cfg file about controlling the idle_timeout.
But in this example, the temperature of the extruder is set to 0, the bed temperature is untouched.

When my printer has stopped with his current job, I will try this without the M104 code....  

```
[idle_timeout]
gcode:
{% if printer.pause_resume.is_paused %}
M104 S0
{% else %}
TURN_OFF_HEATERS
M84
{% endif %}

```

Bully85, 2023-10-21:
M84 disable steper !!!!!!!

lucwuyts, 2023-10-21:
Yes, I don't care about the disable steppers instruction since the printer is possibly in a canceled state.
This addition solved the problem.  (I did erase the line with M104   S0)
I have been away for more than 4 hours, just returned, inserted the fillament again, and pushed resume.   Started as it should.