Bully85/Sovol-SV04-Klipper #35: Unable to connect printer

ginwer7619, 2024-02-12:
Hello, I followed the installation like descriptor but I am unable to connect the printer to Klipper. I always get the error mcu 'mcu': Unable to connect. I tried several cables, reinstalled several times everything but impossible to connect the printer.
I use a Raspberry Pi 4B with 7" TFT. On my Ender3 Neo there is no problem.
Where to find a solution for this connection problem.
Thank you

Bully85, 2024-02-12:
Do you install the mcu for pi?

https://www.klipper3d.org/Measuring_Resonances.html

https://www.klipper3d.org/RPi_microcontroller.html

ginwer7619, 2024-02-12:
Yes, I follow the install, no errors but unable to connect, always the mcu error and no serial device with. Always the message  ls: cannot access '/dev/serial/by-path/*': No such file or directory

Bully85, 2024-02-12:
After flash the firmware rename it from bin to cur?

ginwer7619, 2024-02-13:
Problem solved after isolating VCC pin on USB cable connection was made immediately.
Thanks

Bully85, 2024-02-13:
Thanks for feedback

closed 2024-02-16 by ginwer7619