Bully85/Sovol-SV04-Klipper #36: Kalibrierungswürfel

ginwer7619, 2024-02-14:
Hallo, ich habe den Kalibrierungswürfel gedruckt und eine Abweichung von 3,97mm. Nur finde ich nicht raus wo ich die +3,9 eingeben muss um die Abweichung zu korrigieren. 
Sonst scheint alles zu funktionieren. 
Vielen Dank für Ihre Hilfe. 
Mit freundlichen Grüßen.

Bully85, 2024-02-14:
Es gibt das xy macro da gibst du es ein aber Achtung entweder x oder y sind + und - vertauscht einfach ausprobieren

ginwer7619, 2024-02-14:
Ist das im idex_mode.cfg unter
[gcode_macro DC_VARS]
# Offset coordinates for dual_carriage
variable_offset_x: 0
Gebe ich hier die abweichung ein. En xy macro habe ich nicht.

ginwer7619, 2024-02-15:
Hallo Habe das Problem gelöst, xy macro gefunden. Druck ist jetzt perfekt. Jedoch habe ich ein anderen Problem. Im copy mode läuft alles perfekt bis zum Druckbeginn, der Druck fängt an jedoch ohne aufheizen der Düsen und des Bettes.
Ich benutze Prusaslicer auf MAC, letzte version und das Prusaprofil  aus Sovol-SV04-Klipper-main. Vielen Dank im Voraus

ginwer7619, 2024-02-15:
Hallo, Ja steht drin
;TYPE:Custom
M190 S0
M109 S0
G28 X
SET_DUAL_CARRIAGE CARRIAGE=1 MODE=COPY
START_PRINT EXTRUDER=0 EXTRUDERS=2 EXTRUDER_TEMP=210 EXTRUDER1_TEMP=210 BED_TEMP=60 
AREA_START=29.7633,79.7414 AREA_END=113.206,129.259
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

Bully85, 2024-02-15:
Hast du irgendwas an den cfg Dateien geändert?

ginwer7619, 2024-02-15:
Nein alles übernommen wie runtergeladen. Könnte höchstens Prusa löschen und neu installieren. Dort hatte ich verschiedene änderungen gemacht für meine umgebaute creality. Könnte das Problem davon kommen.

Bully85, 2024-02-15:
Hast du discord? Am besten schickst du mir mal den gcode und ich schau mal. Ich glaube nicht unbedingt dass deine Änderungen bei anderen Druckern hier zu trage kommen aber ausschließen kann ich es nicht wichtig ist zu wissen was im gcode steht.

Hast du auch die aktuellen cfg und prusa Profile aus meinem Release?

ginwer7619, 2024-02-15:
Discord habe ich nicht. cfg und Prusa profile sind as dem release.

Bully85, 2024-02-15:
Was hast du als gcode flavor in prusa

ginwer7619, 2024-02-15:
Habe soeben was im mirror mode versucht, dort klappt es. Werde Prusa neu installieren und morgen nochmal versuchen im copy mode.

Bully85, 2024-02-15:
Ok halt mich auf den Stand

ginwer7619, 2024-02-16:
Hallo, nach Neuinstallation von Prusa klappt es jetzt. Das einzige was ist, die Prime Line ist mir etwas zu weit im Bett. Wo kann ich den Abstand verringern. Ich hatte das normalerweise im G-Code auf 5mm stehen (G1 X5.0 Y20 Z0.3 F5000.0 ; Move to start position
G1 X5.0 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X5.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X5.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line)
ist aber jetzt auf etwa 15mm.
Im Start_End macro steht nur PRIME_LINE habe aber nicht heraus gefunden wo man die Angaben dazu machen kann.

ginwer7619, 2024-02-16:
Habs gefunden, Danke. Da hatte ich natürlich nicht gekuckt. Noch eine Frage zur Anzeige, dort ist die Anzeige der Ventilatoren nicht komplett, ist das auch irgendwo einstellbar.
[image]

Bully85, 2024-02-16:
Im Copy Mode und Miro Mode gibt es keine zwei

ginwer7619, 2024-02-16:
Was bedeutet dann das 2te F: 5. Der Ventilator vom rechten Druckkopf läuft aktuell mit 50% daher die Annahme dass da der Schluss fehlt in der Anzeige.

ginwer7619, 2024-02-16:
Ich habe auch unter Klipper sie Anzeige Unofficial remote url: https://github.com/Bully85/klipper Wie kann ich dies beheben. In Kiauh habe ich die Datei geändert wie Beschrieben.

Bully85, 2024-02-16:
Okay, schade dass du recht hast. Da wird was abgeschnitten sein muss ich heute Abend mal bei mir schauen wie es aussieht

Bully85, 2024-02-16:
Das kannst du beheben, indem du Softrepair machst

ginwer7619, 2024-02-16:
Hallo, noch ein Problem. Ich drucke aktuell nur mit dem rechten Druckkopf. Dabei zeigt der Bildschirm Layer und Filament komplett daneben an. Aktuell müsste er irgendwo um Layer 80 sein. Ist das auch irgendwo zu beheben?
[image]

Bully85, 2024-02-16:
Mit dem Recht im Kopf in Dual Modus oder Rechtsmodus

ginwer7619, 2024-02-22:
Hallo, gibt es ein Problem im copy mode mit Support's. Wenn ich im copy mode drucken möchte mit support drucken beide Köpfe vom ersten Layer den Support und dann geht der linke Kopf in Park Stellung und der rechte fährt auf die linke Druckseite (original) und fängt an zu drucken jedoch ohne Materialvorschub. Beide Düsen und Bett bleiben auf Drucktemperatur. Drucke ich das gleiche Teil zum Test ohne Support geht es. Slicer ist Prusa mit Ihren Einstellungen
Vielen Dank

ginwer7619, 2024-03-16:
> Okay, scheint dass du recht hast. Da wird was abgeschnitten sein muss ich heute Abend mal bei mir schauen wie es aussieht

Hallo, schon was raus gefunden mit den abgeschnittenen Anzeigen?

Bully85, 2024-03-16:
Bin im Moment mit meiner Arbeit sehr eingespannt. Sobald ich Luft finde, schaue ich nach allen Problemen, die bis jetzt auftreten.

ginwer7619, 2024-04-05:
Hallo, ich habe ein Problem nachdem input shapper gemacht wurde. Beim Drucken wird Home und Z-Tilt normal ausgeführt jedoch beim Bed Mesh (Komplet oder Area) kommt nach der 1ten Messstelle eine Fehlermeldung BLTouch failed to deploy danach Failed to verify BLTouch probe is raised; retrying. und nochmal BLTouch failed to deploy und der Drucker bleibt stehen. Nach der ersten Messung zieht Bltouch den Messstift zurück. Klammere ich in Printer.cfg die input-shaper Zeilen aus geht wieder alles normal. Muss da noch was irgendwo geändert werden dass BlTouch keinen Fehler erzeugt? Alle Dateien sind original. Vielen Dank.

ginwer7619, 2024-04-08:
Hallo, Bed Mesh Problem gelöst. Habe probe_with_touch_mode: True unter bltouch eingefügt und schon klappt's.
Mit freundlichen Grüßen.

ginwer7619, 2024-07-25:
> > Okay, scheint dass du recht hast. Da wird was abgeschnitten sein muss ich heute Abend mal bei mir schauen wie es aussieht
> 
> Hallo, schon was raus gefunden mit den abgeschnittenen Anzeigen?

Hallo, Gibts schon was neues betreffend der abgeschnittenen Anzeige (meine Nachricht vom 16/02/24)?

Mit freundlichen Grüßen.

Bully85, 2024-07-25:
Hast du mittlerweile mal Klipperscreen geupdated?

ginwer7619, 2024-07-25:
Hallo,Ja alles auf dem letzten Stand.

closed 2024-10-15 by ginwer7619